aws/aws-fpga #688: Create DCP encounter port missing

zackyen53614, 2025-03-12:
Hello:

I use aws-fpga f1_xdma_shell branch to implement aws_create_dcp_from_tcl flow to get dcp and tar file, and it seems that the flow has finished synthesis and inplementation (also dcp file has been generated).

However, there has a missing port error shown in attachment picture, and caller_combined_wrapper is our design and I think cl_simple_bb may be some cl built-in aws-fpga? I just can't find where are the cl_simple_bb in library work_cl_cl_1 with NOFILE(?) and caller_combined_wrapper.edf

Moreover, the missing port shown are about CLK_300M_DIMM_0_DP, CLK_300M_DIMM_0_DN, ..., and I find these port in common/shell/build/constraint/cl_ddr.xdc constraint file, so I replace these port names with our design DDR I/O port, but run the create_dcp process still has same issue

Thank you!

[image]

[image]

[image]

[image]

czfpga, 2025-03-12:
It seems that the `CL_NAME` wasn't defined in the design so the linker cannot find the correct CL. Can you check if you have it defined? Here is an example: https://github.com/aws/aws-fpga/blob/f1_xdma_shell/hdk/cl/examples/cl_hello_world/design/cl_hello_world_defines.vh#L20 Replace with `caller_combined_wrapper` 

Also,  you need to keep using `cl_ports.vh` for the CL I/O definitions. Redefining the CL/Shell interface is not allowed and will lead to a build failure.

czfpga, 2025-04-28:
I'll close this issue. Please create a new one if you have any new issue. Thanks.

closed 2025-04-28 by czfpga