aws/aws-fpga #689: F2 Getting started guide.

anonsachin, 2025-03-13:
Hey there, I'm a newbie to using cloud FPGAs and looking for a guide that covers both the software and RTL side of things. I've reviewed the F2 docs and successfully ran the `cl_sde` example. But I'm still a bit confused about how things work, especially the interaction from software to RTL, is there a simpler example available?

I noticed in the `cl_sde` example that the `uio` driver is used in [virtual_ethernet_setup.py](https://github.com/aws/aws-fpga/blob/f2/sdk/apps/virtual-ethernet/scripts/virtual_ethernet_setup.py), and then it works with the dpdk driver. I'm trying to figure out how to interact with the RTL from the software—like manipulating registers directly. Ideally, I'd love a basic example, maybe something like adding two numbers or a systolic matrix multiplier, where you send values from the software (using the `uio` driver or `sysfs`) and get the result back.

If there's no such example, I could work with someone from the team to add one to the repo so that other newcomers can also benefit.

anonsachin, 2025-03-17:
Hi @AWSjoeluc ,

Thank you, that was very helpful. The [fpga_pci_sysfs.c](https://github.com/aws/aws-fpga/blob/f7dd0ccb017e490c6a5f0052728ba99b485933f1/sdk/userspace/fpga_libs/fpga_pci/fpga_pci_sysfs.c) file has the implementation using sysfs to get the details of the device and get the `BAR`(Registers) to start interacting with the FPGA. 

I am interested in the [test_dram_dma_hwsw_cosim](https://github.com/aws/aws-fpga/blob/f2/hdk/cl/examples/cl_dram_hbm_dma/software/runtime/test_dram_dma_hwsw_cosim.c), where we are using the `svdpi` to have a mock of the `FPGA` functionality for the sv test and use the sysfs implementation for the hardware test. Is there more documentation on this or a guide? 

Also the `sv_dpi_tasks.svh` only exposes communication to OCL right now [sv_fpga_pci_peek](https://github.com/aws/aws-fpga/blob/f2/hdk/common/verif/include/sh_dpi_tasks.svh#L80), we have to change this in case we want to use other interfaces.

I wanted to understand how to send burst transactions from software to `pcis` and how they will work. For example, if I continuously send  16 32-bit values, will it appear as a 512-bit value? And what would be the address asserted on the axi bus?

AWSjoeluc, 2025-03-17:
> I am interested in the [test_dram_dma_hwsw_cosim](https://github.com/aws/aws-fpga/blob/f2/hdk/cl/examples/cl_dram_hbm_dma/software/runtime/test_dram_dma_hwsw_cosim.c), where we are using the svdpi [...]

> Is there more documentation on the testing framework used, like more details on sh_bfm and things like that?

We do not have extensive documentation on cosim since most hardware workloads consume more bandwidth than is feasible to simulate. You can see these `#if defined(SV_TEST)` branches in [test_dram_dma_hwsw_cosim](https://github.com/aws/aws-fpga/blob/f2/hdk/cl/examples/cl_dram_hbm_dma/software/runtime/test_dram_dma_hwsw_cosim.c#L228-L245). I recommend referring to the [RTL Simulation Guide for HDK Design Flow
](https://github.com/aws/aws-fpga/blob/f2/hdk/docs/RTL_Simulation_Guide_for_HDK_Design_Flow.md) as an introduction.

> Also the sv_dpi_tasks.svh only exposes communication to OCL right now [sv_fpga_pci_peek](https://github.com/aws/aws-fpga/blob/f2/hdk/common/verif/include/sh_dpi_tasks.svh#L80), we have to change this in case we want to use other interfaces.

On `sv_dpi_tasks.svh`, we expect your tests to individually call `tb.card.fpga.sh.peek()` and `tb.card.fpga.sh.poke()` if you're attempting to access anything other than OCL since those interfaces behave differently.

> I wanted to understand how to send burst transactions from software to `pcis` and how they will work. For example, if I continuously send 16 32-bit values, will it appear as a 512-bit value? And what would be the address asserted on the axi bus?

The SDK offers two methods:

1. [fpga_dma_burst_read() & fpga_dma_burst_write()](https://github.com/aws/aws-fpga/blob/f7dd0ccb017e490c6a5f0052728ba99b485933f1/sdk/userspace/fpga_libs/fpga_dma/fpga_dma_utils.c#L130-L176)
     * You can see them reference when using [CL_DRAM_HBM_DMA with the XDMA shell](https://github.com/aws/aws-fpga/blob/f2/hdk/cl/examples/cl_dram_hbm_dma/software/runtime/test_dram_dma_hwsw_cosim.c#L228-L245
) (being released in late Q2 2025) 
3. [fpga_pci_write_burst](https://github.com/aws/aws-fpga/blob/f7dd0ccb017e490c6a5f0052728ba99b485933f1/sdk/userspace/fpga_libs/fpga_pci/fpga_pci.c#L435-L452)
Otherwise I recommend

anonsachin, 2025-03-18:
> We do not have extensive documentation on cosim since most hardware workloads consume more bandwidth than is feasible to simulate. You can see these `#if defined(SV_TEST)` branches in [...]
I Have gone through this and set up a simple adder circuit with the `ocl` bus and have simulated using the regular `tb.poke` interface. I will try to use [test_dram_dma_hwsw_cosim](https://github.com/aws/aws-fpga/blob/f2/hdk/cl/examples/cl_dram_hbm_dma/software/runtime/test_dram_dma_hwsw_cosim.c#L228-L245) as a reference and create a cosim. You won't always need to do a full throughput test, right? We can do that next on HW. This will help quantify and set up APIs for behavioural testing that is not at full throughput.

> On `sv_dpi_tasks.svh`, we expect your tests to individually call `tb.card.fpga.sh.peek()` and `tb.card.fpga.sh.poke()` if you're attempting to access anything other than OCL since those interfaces behave differently.
> 
Okay, will do that.
> The SDK offers two methods:
> 
> 1. [fpga_dma_burst_read() & fpga_dma_burst_write()](https://github.com/aws/aws-fpga/blob/f7dd0ccb017e490c6a5f0052728ba99b485933f1/sdk/userspace/fpga_libs/fpga_dma/fpga_dma_utils.c#L130-L176)
>    
>    * You can see them reference when using [CL_DRAM_HBM_DMA with the XDMA shell](https://github.com/aws/aws-fpga/blob/f2/hdk/cl/examples/cl_dram_hbm_dma/software/runtime/test_dram_dma_hwsw_cosim.c#L228-L245) (being released in late Q2 2025)
> 2. [fpga_pci_write_burst](https://github.com/aws/aws-fpga/blob/f7dd0ccb017e490c6a5f0052728ba99b485933f1/sdk/userspace/fpga_libs/fpga_pci/fpga_pci.c#L435-L452)
>    Otherwise I recommend

`fpga_pci_write_burst` - can I replicate this in sv test with [dma_buffer_to_cl](https://github.com/aws/aws-fpga/blob/f7dd0ccb017e490c6a5f0052728ba99b485933f1/hdk/common/verif/models/sh_bfm/sh_bfm.sv#L1806) and `sv_host_memory` or is there another way?

AWSjoeluc, 2025-03-18:
Yes, [dma_buffer_to_cl](https://github.com/aws/aws-fpga/blob/f7dd0ccb017e490c6a5f0052728ba99b485933f1/hdk/common/verif/models/sh_bfm/sh_bfm.sv#L1806) and sv_host_memory should be the correct way to perform this in the test bench. Happy to hear the recommendations above will help!