aws/aws-fpga #692: 2024.2 Not Supported

kaasaraai-jump, 2025-03-18:
I'm using the following AMI, it comes with vivado 2024.2, but the github doesn't seem to be supporting it, log below.  Is there a different AMI I should use to get 2024.1?

`FPGA Developer AMI (Ubuntu) - 1.17.0-prod-yexmzwy3ipomw-ami-05c9eff928992ede5.4-prod-byisb4uqt2pwc`

```
kaasaraai@ause1a-ossdev-firedancer1:~/aws-fpga$ source hdk_setup.sh -d
INFO: Setting up environment variables
INFO: Changing AWS_FPGA_REPO_DIR from /home/ubuntu//src/project_data//aws-fpga to /home/kaasaraai/aws-fpga
DEBUG: Checking for Vivado install:
INFO: Using vivado v2024.2 (64-bit)
ERROR: vivado v2024.2 (64-bit) is not supported by this HDK release.
ERROR: Supported versions are:
vivado v2024.1 (64-bit)
```

mjthimm, 2025-03-18:
Hello!

You are correct! Please use FPGA Developer AMI (Ubuntu) - 1.16.0 for the moment: https://aws.amazon.com/marketplace/pp/prodview-f5kjsenkfkz5u

mjthimm, 2025-03-18:
To investigate this further, could you please provide me with your AWS account ID number?

mjthimm, 2025-03-19:
Sounds good! Please also file a support ticket for this request: https://docs.aws.amazon.com/awssupport/latest/user/case-management.html#creating-a-support-case

closed 2025-04-14 by AWSjoeluc