aws/aws-fpga #694: F2 cl_mem_perf example failures

pyz-creeper, 2025-03-21:
Hi, I am working on mem_perf example and I encountered several test failures:
```
ubuntu@ip-172-31-13-192:~/aws-fpga/hdk/cl/examples/cl_mem_perf/software/runtime$ ./test_dram_hbm_dma
2025-03-21T13:40:05.403944Z, test_dram_hbm_dma, ERROR, test_dram_hbm_dma.c +123: dma_example(): unable to open read dma queue
2025-03-21T13:40:05.404000Z, test_dram_hbm_dma, ERROR, test_dram_hbm_dma.c +83: main(): DMA example failed
2025-03-21T13:40:05.404011Z, test_dram_hbm_dma, INFO, test_dram_hbm_dma.c +96: main(): TEST FAILED
ubuntu@ip-172-31-13-192:~/aws-fpga/hdk/cl/examples/cl_mem_perf/software/runtime$ ./test_aws_clk_gen
===================================================
Running test_aws_clk_gen 
===================================================
slot_id     = 0
===================================================
__INFO__: deassert_clk_resets()
2025-03-21T13:40:27.691568Z, test_aws_clk_gen, ERROR, test_dram_dma_common.c +208: deassert_clk_resets(): Unable to attach to the AFI on slot id 0
2025-03-21T13:40:27.691596Z, test_aws_clk_gen, ERROR, test_aws_clk_gen.c +109: main(): Failed deassert_clk_resets()
2025-03-21T13:40:27.691598Z, test_aws_clk_gen, INFO, test_aws_clk_gen.c +118: main(): TEST FAILED
ubuntu@ip-172-31-13-192:~/aws-fpga/hdk/cl/examples/cl_mem_perf/software/runtime$ ./test_clk_freq
===================================================
Running test_clk_freq 
===================================================
slot_id     = 0
===================================================
__INFO__: Dynamic config clk_extra_a1 = 87 MHz, clk_extra_b0 = 125 MHz, clk_extra_c0 = 250 MHz, clk_hbm_axi = 437 MHz
2025-03-21T13:40:43.362993Z, test_clk_freq, ERROR, test_clk_freq.c +119: main(): aws_clkgen_set_dynamic()
2025-03-21T13:40:43.363019Z, test_clk_freq, INFO, test_clk_freq.c +138: main(): TEST FAILED
ubuntu@ip-172-31-13-192:~/aws-fpga/hdk/cl/examples/cl_mem_perf/software/runtime$ ./test_hbm_perf32 
===================================================
Running test_hbm_perf32 
===================================================
slot_id     = 0
cfg_axlen   = 0x0000000f
cfg_wdata   = 0x12345678
cfg_wr_ctl  = 0xffffffff
cfg_rd_ctl  = 0xffffffff
cfg_runtime = 0x0000001e
===================================================
__INFO__: deassert_clk_resets()

2025-03-21T13:41:40.779068Z, test_hbm_perf32, ERROR, test_dram_dma_common.c +208: deassert_clk_resets(): Unable to attach to the AFI on slot id 0
2025-03-21T13:41:40.779096Z, test_hbm_perf32, ERROR, test_hbm_perf32.c +136: main(): Failed deassert_clk_resets()
2025-03-21T13:41:40.779099Z, test_hbm_perf32, INFO, test_hbm_perf32.c +157: main(): TEST FAILED
```
As [cl_mem_perf](https://github.com/aws/aws-fpga/blob/f2/hdk/cl/examples/cl_mem_perf/README.md) README says, features using XDMA engine are currently unsupported. Is it fair to say that any test related to dma will fail on F2 instances?
If I want to transfer large amount of data from host to FPGA, my **only solution now** is to write a dpdk application like dpdk-testpmd, which send/recv packets through virtual ethernet?

czfpga, 2025-03-21:
Hi @pyz-creeper,

In addtion, what test AFI did you use to run those runtime tests? 
That's correct. Tests that require the DMA engine are currently disabled until the XDMA shell support is added, which will be coming very soon. The team is also planning to release more SDE examples to demonstrate H2C and C2H data transactions. Please stay tuned on both. For now, please continue referring to the DPDK PMD example.

pyz-creeper, 2025-03-22:
Thank you @czfpga. I am using a AFI build from cl_mem_perf's hardware code. (I think I am building a correct AFI, because that explain the failure?)
I will write a DPDK program to substitute XDMA engine for large amount IO.

closed 2025-03-22 by pyz-creeper