aws/aws-fpga #695: Pblock 'pblock_CL_SLR1_XBAR' is not contained by the parent Pblock 'pblock_CL_SLR1'

monological, 2025-03-27:
I’m trying to synthesize my design, but DRC fails with the error below while running opt_design in build_level_1_cl.tcl. I believe there may be an issue with the AWS-provided constraints file, small_shell_cl_pnr_user.xdc — specifically in the following section:

```
create_pblock pblock_CL_SLR1_XBAR

# Complete CRs in SLR1
resize_pblock pblock_CL_SLR1_XBAR -add {CLOCKREGION_X0Y4:CLOCKREGION_X4Y7}

# Partial CRs
set_property parent pblock_CL_SLR1 [get_pblocks pblock_CL_SLR1_XBAR]

########################################
# Module Mapping
########################################
add_cells_to_pblock pblock_CL_SLR1_XBAR [get_cells [list WRAPPER/CL/CL_DMA_PCIS_SLV/AXI4_CROSSBAR ]]
```

However, this same constraints file is used in the cl_dram_hbm_dma example, which builds without any issues, so I’m at a loss as to why it would fail like that.

The only addition to the xdc is adding my design's cells to the pblocks:

```
add_cells_to_pblock [get_pblocks pblock_CL_SLR0] [get_cells -hierarchical -filter {NAME =~ WRAPPER_INST/CL/top_inst/sha512_pre_inst*}]
add_cells_to_pblock [get_pblocks pblock_CL_SLR0] [get_cells -hierarchical -filter {NAME =~ WRAPPER_INST/CL/top_inst/pad_th_inst*}]

add_cells_to_pblock [get_pblocks pblock_CL_SLR0] [get_cells -hierarchical -filter {NAME =~ WRAPPER_INST/CL/top_inst/sha_f_inst*}]
add_cells_to_pblock [get_pblocks pblock_CL_SLR0] [get_cells -hierarchical -filter {NAME =~ WRAPPER_INST/CL/top_inst/sha512_modq_meta_inst*}]
add_cells_to_pblock [get_pblocks pblock_CL_SLR0] [get_cells -hierarchical -filter {NAME =~ WRAPPER_INST/CL/top_inst/sha_th_inst*}]

add_cells_to_pblock [get_pblocks pblock_CL_SLR1] [get_cells -hierarchical -filter {NAME =~ WRAPPER_INST/CL/top_inst/sv0_f_inst*}]
add_cells_to_pblock [get_pblocks pblock_CL_SLR1] [get_cells -hierarchical -filter {NAME =~ WRAPPER_INST/CL/top_inst/ed25519_sigverify_0_inst*}]
add_cells_to_pblock [get_pblocks pblock_CL_SLR1] [get_cells -hierarchical -filter {NAME =~ WRAPPER_INST/CL/top_inst/sv0_th_inst*}]

add_cells_to_pblock [get_pblocks pblock_CL_SLR2] [get_cells -hierarchical -filter {NAME =~ WRAPPER_INST/CL/top_inst/sv1_f_inst*}]
add_cells_to_pblock [get_pblocks pblock_CL_SLR2] [get_cells -hierarchical -filter {NAME =~ WRAPPER_INST/CL/top_inst/ed25519_sigverify_1_inst*}]
add_cells_to_pblock [get_pblocks pblock_CL_SLR2] [get_cells -hierarchical -filter {NAME =~ WRAPPER_INST/CL/top_inst/sv1_th_inst*}]

add_cells_to_pblock [get_pblocks pblock_CL_SLR2] [get_cells -hierarchical -filter {NAME =~ WRAPPER_INST/CL/top_inst/sv2_i_pipe_inst*}]
add_cells_to_pblock [get_pblocks pblock_CL_SLR2] [get_cells -hierarchical -filter {NAME =~ WRAPPER_INST/CL/top_inst/sv2_f_inst*}]
add_cells_to_pblock [get_pblocks pblock_CL_SLR2] [get_cells -hierarchical -filter {NAME =~ WRAPPER_INST/CL/top_inst/ed25519_sigverify_2_inst*}]
```

---

Built using aws-fpga (v2.0.5) (f2 branch)

**log file:** [2025_03_26-154610.vivado.log](https://github.com/user-attachments/files/19489226/2025_03_26-154610.vivado.log)

**The error:**

> ERROR: [DRC HDPR-23] Nested Pblock ranges must be a subset of parent Pblock ranges: The child Pblock 'pblock_CL_SLR1_XBAR' is not contained by the parent Pblock 'pblock_CL_SLR1'.
> Child ranges 'CLOCKREGION_X0Y4:CLOCKREGION_X4Y7 '.
> Parent ranges 'URAM288_X0Y64:URAM288_X3Y127 RIU_OR_X0Y24:RIU_OR_X0Y31 RIU_OR_X0Y16:RIU_OR_X0Y19 RAMB36_X9Y48:RAMB36_X10Y95 RAMB36_X8Y72:RAMB36_X8Y95 RAMB36_X8Y48:RAMB36_X8Y59 RAMB36_X0Y48:RAMB36_X7Y95 RAMB18_X9Y96:RAMB18_X10Y191 RAMB18_X8Y144:RAMB18_X8Y191 RAMB18_X8Y96:RAMB18_X8Y119 RAMB18_X0Y96:RAMB18_X7Y191 PLL_X0Y12:PLL_X0Y15 PLL_X0Y8:PLL_X0Y9 MMCM_X0Y6:MMCM_X0Y7 MMCM_X0Y4:MMCM_X0Y4 IOB_X0Y312:IOB_X0Y415 IOB_X0Y208:IOB_X0Y259 ILKNE4_X0Y0:ILKNE4_X0Y0 HPIO_VREF_SITE_X0Y12:HPIO_VREF_SITE_X0Y15 HPIO_VREF_SITE_X0Y8:HPIO_VREF_SITE_X0Y9 HPIO_RCLK_PRBS_X0Y6:HPIO_RCLK_PRBS_X0Y7 HPIO_RCLK_PRBS_X0Y4:HPIO_RCLK_PRBS_X0Y4 HPIOB_DCI_SNGL_X0Y24:HPIOB_DCI_SNGL_X0Y31 HPIOB_DCI_SNGL_X0Y16:HPIOB_DCI_SNGL_X0Y19 HPIOBDIFFOUTBUF_X0Y144:HPIOBDIFFOUTBUF_X0Y191 HPIOBDIFFOUTBUF_X0Y96:HPIOBDIFFOUTBUF_X0Y119 HPIOBDIFFINBUF_X0Y144:HPIOBDIFFINBUF_X0Y191 HPIOBDIFFINBUF_X0Y96:HPIOBDIFFINBUF_X0Y119 HARD_SYNC_X18Y8:HARD_SYNC_X21Y15 HARD_SYNC_X16Y12:HARD_SYNC_X17Y15 HARD_SYNC_X16Y8:HARD_SYNC_X17Y9 HARD_SYNC_X0Y8:HARD_SYNC_X15Y15 GTYE4_COMMON_X0Y4:GTYE4_COMMON_X0Y7 GTYE4_CHANNEL_X0Y16:GTYE4_CHANNEL_X0Y31 DSP48E2_X17Y90:DSP48E2_X24Y185 DSP48E2_X16Y138:DSP48E2_X16Y185 DSP48E2_X16Y90:DSP48E2_X16Y113 DSP48E2_X0Y90:DSP48E2_X15Y185 CMACE4_X0Y2:CMACE4_X0Y4 BUFG_GT_SYNC_X0Y60:BUFG_GT_SYNC_X0Y119 BUFG_GT_X0Y96:BUFG_GT_X0Y191 BUFGCTRL_X0Y48:BUFGCTRL_X0Y63 BUFGCTRL_X0Y32:BUFGCTRL_X0Y39 BUFGCE_DIV_X0Y24:BUFGCE_DIV_X0Y31 BUFGCE_DIV_X0Y16:BUFGCE_DIV_X0Y19 BUFGCE_X0Y144:BUFGCE_X0Y191 BUFGCE_X0Y96:BUFGCE_X0Y119 BITSLICE_TX_X0Y48:BITSLICE_TX_X0Y63 BITSLICE_TX_X0Y32:BITSLICE_TX_X0Y39 BITSLICE_RX_TX_X0Y312:BITSLICE_RX_TX_X0Y415 BITSLICE_RX_TX_X0Y208:BITSLICE_RX_TX_X0Y259 BITSLICE_CONTROL_X0Y48:BITSLICE_CONTROL_X0Y63 BITSLICE_CONTROL_X0Y32:BITSLICE_CONTROL_X0Y39 BIAS_X0Y12:BIAS_X0Y15 BIAS_X0Y8:BIAS_X0Y9 SLICE_X166Y240:SLICE_X175Y479 SLICE_X164Y300:SLICE_X165Y419 SLICE_X154Y240:SLICE_X163Y479 SLICE_X152Y300:SLICE_X153Y419 SLICE_X141Y240:SLICE_X151Y479 SLICE_X139Y300:SLICE_X140Y419 SLICE_X126Y240:SLICE_X138Y479 SLICE_X124Y300:SLICE_X125Y419 SLICE_X122Y240:SLICE_X123Y479 SLICE_X117Y360:SLICE_X121Y479 SLICE_X117Y240:SLICE_X121Y299 SLICE_X113Y240:SLICE_X116Y479 SLICE_X111Y300:SLICE_X112Y419 SLICE_X99Y240:SLICE_X110Y479 SLICE_X97Y300:SLICE_X98Y419 SLICE_X87Y240:SLICE_X96Y479 SLICE_X85Y300:SLICE_X86Y419 SLICE_X65Y240:SLICE_X84Y479 SLICE_X63Y300:SLICE_X64Y419 SLICE_X52Y240:SLICE_X62Y479 SLICE_X50Y300:SLICE_X51Y419 SLICE_X39Y240:SLICE_X49Y479 SLICE_X37Y300:SLICE_X38Y419 SLICE_X21Y240:SLICE_X36Y479 SLICE_X19Y300:SLICE_X20Y419 SLICE_X10Y240:SLICE_X18Y479 SLICE_X8Y300:SLICE_X9Y419 SLICE_X0Y240:SLICE_X7Y479 '.
> Resolution: add the missing ranges to the parent or child Pblock.
> INFO: [Project 1-461] DRC finished with 1 Errors

rcko-aws, 2025-03-27:
Hello, did you check in Vivado floorplanning view if your new pblock is indeed inside of the parent?

rcko-aws, 2025-03-31:
Currently recommend not creating pblock_CL_SLR1_XBAR and placing those cells in  pblock_CL_SLR1
The irregular pblock_CL_SLR1 does not completely overlay CLOCKREGION_X0Y4:CLOCKREGION_X4Y7 as DRC points out.
We are investigating why its passing on our end

monological, 2025-03-31:
I ended up removing it and got it to work. I've attached the final xdc and a diff that worked for me.

[small_shell_cl_pnr_user.xdc.diff.txt](https://github.com/user-attachments/files/19538120/small_shell_cl_pnr_user.xdc.diff.txt)

[small_shell_cl_pnr_user.xdc.txt](https://github.com/user-attachments/files/19538084/small_shell_cl_pnr_user.xdc.txt)

czfpga, 2025-03-31:
@monological,

Thanks for reaching out. The `pblock_CL_SLR1_XBAR` FP constraint is correct. However, it's relying on some default Vivado tool behavior that might not be very explicitly clear at first sight. As described in [this AMD UG](https://docs.amd.com/r/en-US/ug909-vivado-partial-reconfiguration/Floorplanning-for-Versal-Devices) on SNAPPING_MODE, the `pblock_CL_SLR1_XBAR` is using the `NESTED` snapping mode, which is the default for any child pblock whose parent pblock's snapping mode is set to `ON`. This approach simplifies the user's pblock definition, especially when working with a parent pblock region that doesn't have a perfect rectangular shape. Hope that answers your question.

Thanks,

Chen

closed 2025-04-14 by AWSjoeluc