aws/aws-fpga #710: Simulation errors

pyz-creeper, 2025-05-01:
I am trying to integrate HBM into sde engines to allow large amount data stored on my FPGA. But after some simulation, I found that I can not access HBM module with axi4 (all read requests does not receive a valid). I started to run simulations in cl_mem_perf and cl_dram_hbm_dma to get a ground truth axi4 usage.

Error 1: cl is not correct.
I found that some simulations in cl_mem_perf and cl_dram_hbm_dma examples are not correctly executed. I dumped the waves by adding "log_wave -r /" in waves.tcl, and I found the cl module is still cl_sde (while it should be cl_mem_perf or cl_dram_hbm_dma). It seems that the simulation is executed on a completely wrong cl.

I run my simulations like this:
cd aws-fpga
source hdk_setup.sh
cd hdk/cl/examples/cl_mem_perf
export CL_DIR=$(pwd)
cd verif/scripts
make TEST=test_hbm_perf32 > perf32.log

Error2: xvlog runs for more than 1 hour
Normally, my simulation takes about 2 hours, and the overhead is on xelab (the simulation part). But, when I simulate some test examples in cl_mem_perf, xvlog execution takes more than 1 hour and still does not finish.
For example, this xvlog command runs for more than 1 hour:
cd /home/ubuntu/aws-fpga/hdk/cl/examples/cl_mem_perf/verif/sim/xsim/test_hbm_perf32_sv && xvlog --sv -m64 --define DMA_TEST   --define plusarg_save  --define timeout=1ms  --define AWS_SIM --define USE_64GB_DDR_DIMM=1 --work xil_defaultlib --incr --relax -f /home/ubuntu/aws-fpga/hdk/cl/examples/cl_mem_perf/verif/scripts/top.xsim.f
Sometimes, even test_null's xvlog command also has this error. I have to mannualy ctrl+c to stop them. But ctrl+c does not fix this, the next simulation still has this problem. All I can do is go to ec2 console, stop the m4.2xlarge instance I am using, and restart after waiting for several minutes.

I hope you can help me fix these two errors or identify the causes. That will be extremely helpful because these two problems are really slowing me down.

czfpga, 2025-05-01:
Hi @pyz-creeper,

Please ensure you're following the steps in the [RTL simulation guide](https://gitlab.fpga.annapurna.aws.dev/f2/aws-fpga-preview-staging/-/blob/main/hdk/docs/RTL_Simulation_Guide_for_HDK_Design_Flow.md?ref_type=heads#try-out-one-of-hdk-examples-or-write-your-own) correctly to run the example's simulation.

The slow simulation time of the HBM IP is a known issue. Please refer to #5 in [errata](https://github.com/aws/aws-fpga/blob/f2/ERRATA.md). The team has provided this feedback to AMD.

pyz-creeper, 2025-05-03:
Thank you @czfpga , I think the simulation on wrong cl is related to the ip file generation. I solved my problems for now

closed 2025-05-03 by pyz-creeper