aws/aws-fpga #711: Very slow read performance using custom fpga_pci_read_burst implementation (100x slower than write)

pyz-creeper, 2025-05-05:
Hi, I'm using the `cl_mem_perf` example to test direct PCIe interaction between the host and FPGA card. Something unexpected happened during my experiments.

I'm using the `fpga_pci_write_burst` function from `sdk/userspace/fpga_libs/fpga_pci/fpga_pci.c`, and it achieves decent performance (around 2.5 GB/s), which seems reasonable.

To test the read path, I implemented a function called `fpga_pci_read_burst` based on `fpga_pci_write_burst`. Here's the code:
```c
int fpga_pci_read_burst(pci_bar_handle_t handle, uint64_t offset, uint32_t* datap, uint64_t dword_len) {
	uint64_t i;
	log_debug("handle=%d, offset=0x%" PRIx64, handle, offset);

	/** get the pointer to the beginning of the range */
	uint32_t *reg_ptr = (uint32_t *)fpga_pci_bar_get_mem_at_offset(handle,
			offset, sizeof(uint32_t)*dword_len);
	fail_on(!reg_ptr, err, "fpga_plat_get_mem_at_offset failed");

	/** memcpy */
	for (i = 0; i < dword_len; ++i) {
		// reg_ptr[i] = datap[i]; I removed the copy from H2C, add a copy from C2H.
		datap[i] = reg_ptr[i];
	}

	return 0;
err:
	return -EINVAL;
}
```
Functionally, this read implementation appears correct, but the performance is extremely poor. Reading 300 MB of data takes around 100 seconds — which is roughly **1000 times slower** than the write speed. This huge gap really confuses me.
Am I using the wrong approach for reading via PCIe? Is there a more appropriate function I should be using for high-throughput reads?

Thanks in advance!

czfpga, 2025-05-06:
Hi @pyz-creeper,

Here are 2 options for high performance transitions between the host and the card 

1. You could consider migrating to the XDMA shell after it's released and use the XDMA driver for high bandwidth data transfers.
2. You can consider integrating the SDE engine in your design. There is [comprehensive list of APIs and examples](https://github.com/aws/aws-fpga/blob/f2/hdk/cl/examples/cl_sde/software/src/README.md) provided to demonstrate its usage.

There is not enough information on data destination, BAR used, data length, etc, for us to comment on the performance.  In general the write performance is expected to be better than the read performance. You would need to queue up enough outstanding transactions in order to fully utilize the bandwidth of the interface. It's unlikely you can achieve that using the peek and poke type of transactions you're testing with.

pyz-creeper, 2025-05-08:
Hi @czfpga 
> There is not enough information on data destination, BAR used, data length, etc, for us to comment on the performance.

Thanks for the feedback. In my experiment, I used the default settings. The data length I choose is 256MB, and the data destination is HBM. As far as I recall, BAR4 of PF0 (P0-BAR4) is typically used for PCIe operations targeting either DDR or HBM. I never change it.
Is there any other information I can provide to help you determine whether something might be going wrong?

czfpga, 2025-05-08:
Hi @pyz-creeper,

Please consider adding ILA cores to monitor the PCIS interface (BAR4) in the example. It would help identify the source of delay causing the low read bandwidth. Hope that helps.

closed 2025-05-12 by pyz-creeper