aws/aws-fpga #712: Couple AFI to an AMI on the Marketplace

Tuditi, 2025-05-19:
Hi there,

We released [an AMI](https://aws.amazon.com/marketplace/pp/prodview-imfiyzy7svjgu) on the public marketplace, that leverages a specific AFI. Within our organization, we always have access to the AFI if we launch this AMI.

If I launch it as an external user, the AFI isn't loaded due to permission issues. We don't want to make our AFI completely public, since users could bypass our marketplace product easily. Do you have an idea on how to tie the AFI to the given AMI that was purchased through the Marketplace?

mjthimm, 2025-05-20:
Hello,

Did you follow the procedure outlined in our FAQ here: https://github.com/aws/aws-fpga/blob/master/FAQs.md#marketplace-faqs?

If you have,  we recommend contacting Marketplace again and bringing the AFI:AMI link problem you're encountering to their attention.

Best,
Max

Tuditi, 2025-05-20:
Thanks Max,

I reached out to Mukhethwa M. to look into our issue. I followed the procedure outlined in your document (see image). The product was created as a server product, so I didn't see a way to tie the AFI to the AMI. Should this normally happen within the PLF?

[image]

Cheers,
David

mjthimm, 2025-05-20:
Hi David,

When looking to use an AFI in a product, it must be tied to an AMI. It sounds like Marketplace has some digging to do here.

They should have notified you about tying the AFI to the AMI, as that's the standard procedure for how AFIs are bundled into a Marketplace product.

Please let me know what you learn! We're constantly working to improve F2 for yourself and all customers and making your products available is a big part of that.

Best,
Max

Tuditi, 2025-05-21:
Hi Max,

Our quick fix was to make the AFI public. For future listings we'll contact the Marketplace team when to tie AMI's to private AFI's for IP protection. There is an AFI field in the Product Load Form (= excel sheet to be uploaded), but the Marketplace form unfortunately lacks this option. Below you can find the e-mail of the support engineer assigned to our case.

> 
> Good Day,
> 
> Thank you for your response. For future purposes, we suggest that you log a inquiry with our team after listing your product to a limited state. We will work with engineering to complete your request.

Cheers,
David

mjthimm, 2025-05-21:
Hi David,

We'll be taking your experience with this process and working with Marketplace to improve it.

Thank you for your patience on this!

Best,
Max

mjthimm, 2025-06-09:
Hi David,

Were you ever able to get help from Marketplace getting this issue resolved?

If so, do you have a ticket number you could share with me so I can make sure that Marketplace makes this right for you?

Best,
Max