aws/aws-fpga #713: Vitis AFI creation on F2 instance

jkato-vectology, 2025-05-20:
Hi AWS FPGA Team,

Do you have plan for support Vitis AFI creation on f2 instance?
Is there any way to solve this problem?

Thank you for your time and support.

TrahStevAmz, 2025-05-20:
Hello,

Delivering support for Vitis AfI creation on F2 is a very high priority for this team. When we have a release candidate or beta branch to try out, we will let you know.

Thanks,
Trahstev

ywaka-vec, 2025-05-21:
Hi,

I also have a same question, so it's very helpful!

Do you happen to have an estimated timeline for when the release candidate or beta branch might be available? Even a rough idea would be really helpful for our planning.

Thanks.

ywaka-vec, 2025-05-22:
Hi Max,

Thank you for the update.
Understood on the status—I'm looking forward to the release of Vitis AFI generation.
Appreciate all the hard work from you and the team!

Best regards,
ywaka