aws/aws-fpga #723: Want to create simple CL on F2, but running into significant issues

dustydude42, 2025-06-02:
Hi, I am trying to run a small module with 5 io ports on an F2 as a test. The CL_TEMPLATE did not give enough background on how to formulate PCIe/AXI4 communication and the set up for the constraint files.

I have been trying to build off of the cl_dram_hbm_dma example as a framework to interact with my simple io module. I have attempted to hook onto the OCL slave line to use as io for my CL, and I finally got my design files to synthesize, but now I am running into the following issue in implementation.

I thought I should not change the cl_dram_hbm_dma constraint files since my module is not requiring any more ports than what already exists. From the errors below, can anyone tell me what may be wrong? 

Thank you!

WARNING: [Vivado 12-180] No cells matched 'WRAPPER/CL/CL_DMA_PCIS_SLV/SLR0_PIPE_RST_N'. [/home/dev/aws-fpga/hdk/cl/examples/cl_dram_hbm_dma/build/constraints/small_shell_cl_pnr_user.xdc:157]
WARNING: [Vivado 12-180] No cells matched 'WRAPPER/CL/CL_DMA_PCIS_SLV/AXI4_REG_SLC_DDRB_SLR0'. [/home/dev/aws-fpga/hdk/cl/examples/cl_dram_hbm_dma/build/constraints/small_shell_cl_pnr_user.xdc:157]
WARNING: [Vivado 12-180] No cells matched 'WRAPPER/CL/CL_DMA_PCIS_SLV/CL_TST_DDR_B'. [/home/dev/aws-fpga/hdk/cl/examples/cl_dram_hbm_dma/build/constraints/small_shell_cl_pnr_user.xdc:157]
WARNING: [Vivado 12-180] No cells matched 'WRAPPER/CL/CL_HBM'. [/home/dev/aws-fpga/hdk/cl/examples/cl_dram_hbm_dma/build/constraints/small_shell_cl_pnr_user.xdc:157]
CRITICAL WARNING: [Vivado 12-1433] Expecting a non-empty list of cells to be added to the pblock.  Please verify the correctness of the <cells> argument. [/home/dev/aws-fpga/hdk/cl/examples/cl_dram_hbm_dma/build/constraints/small_shell_cl_pnr_user.xdc:157]
Finished Parsing XDC File [/home/dev/aws-fpga/hdk/cl/examples/cl_dram_hbm_dma/build/constraints/small_shell_cl_pnr_user.xdc]
read_xdc: Time (s): cpu = 00:00:38 ; elapsed = 00:00:38 . Memory (MB): peak = 6982.004 ; gain = 0.000 ; free physical = 12433 ; free virtual = 29196
## source ${HDK_SHELL_DIR}/build/scripts/ddr_io_train.tcl
### set DDR_CELL [get_cells WRAPPER/CL/SH_DDR/genblk1.IS_DDR_PRESENT.DDR4_0]
### if {$DDR_CELL != ""} {
###     set_property MIG_FLOORPLAN_MODE FULL [get_cells WRAPPER/CL/SH_DDR/genblk1.IS_DDR_PRESENT.DDR4_0]
### } else {
###     print "Skipped ddr_io_train.tcl. No DDR Core in the design"
### }
## print "Start optimizing customer design ${CL}"

AWS FPGA: (22:00:08): Start optimizing customer design cl_dram_hbm_dma

## opt_design
Command: opt_design
Attempting to get a license for feature 'Implementation' and/or device 'xcvu47p'
INFO: [Common 17-349] Got license for feature 'Implementation' and/or device 'xcvu47p'
Running DRC as a precondition to command opt_design

Starting DRC Task
INFO: [DRC 23-27] Running DRC with 8 threads
ERROR: [DRC HDPR-23] Nested Pblock ranges must be a subset of parent Pblock ranges: The child Pblock 'pblock_CL_SLR1_XBAR' is not contained by the parent Pblock 'pblock_CL_SLR1'.
Child ranges 'CLOCKREGION_X0Y4:CLOCKREGION_X4Y7 '.
Parent ranges 'URAM288_X0Y64:URAM288_X3Y127 RIU_OR_X0Y24:RIU_OR_X0Y31 RIU_OR_X0Y16:RIU_OR_X0Y19 RAMB36_X9Y48:RAMB36_X10Y95 RAMB36_X8Y72:RAMB36_X8Y95 RAMB36_X8Y48:RAMB36_X8Y59 RAMB36_X0Y48:RAMB36_X7Y95 RAMB18_X9Y96:RAMB18_X10Y191 RAMB18_X8Y144:RAMB18_X8Y191 RAMB18_X8Y96:RAMB18_X8Y119 RAMB18_X0Y96:RAMB18_X7Y191 PLL_X0Y12:PLL_X0Y15 PLL_X0Y8:PLL_X0Y9 MMCM_X0Y6:MMCM_X0Y7 MMCM_X0Y4:MMCM_X0Y4 IOB_X0Y312:IOB_X0Y415 IOB_X0Y208:IOB_X0Y259 ILKNE4_X0Y0:ILKNE4_X0Y0 HPIO_VREF_SITE_X0Y12:HPIO_VREF_SITE_X0Y15 HPIO_VREF_SITE_X0Y8:HPIO_VREF_SITE_X0Y9 HPIO_RCLK_PRBS_X0Y6:HPIO_RCLK_PRBS_X0Y7 HPIO_RCLK_PRBS_X0Y4:HPIO_RCLK_PRBS_X0Y4 HPIOB_DCI_SNGL_X0Y24:HPIOB_DCI_SNGL_X0Y31 HPIOB_DCI_SNGL_X0Y16:HPIOB_DCI_SNGL_X0Y19 HPIOBDIFFOUTBUF_X0Y144:HPIOBDIFFOUTBUF_X0Y191 HPIOBDIFFOUTBUF_X0Y96:HPIOBDIFFOUTBUF_X0Y119 HPIOBDIFFINBUF_X0Y144:HPIOBDIFFINBUF_X0Y191 HPIOBDIFFINBUF_X0Y96:HPIOBDIFFINBUF_X0Y119 HARD_SYNC_X18Y8:HARD_SYNC_X21Y15 HARD_SYNC_X16Y12:HARD_SYNC_X17Y15 HARD_SYNC_X16Y8:HARD_SYNC_X17Y9 HARD_SYNC_X0Y8:HARD_SYNC_X15Y15 GTYE4_COMMON_X0Y4:GTYE4_COMMON_X0Y7 GTYE4_CHANNEL_X0Y16:GTYE4_CHANNEL_X0Y31 DSP48E2_X17Y90:DSP48E2_X24Y185 DSP48E2_X16Y138:DSP48E2_X16Y185 DSP48E2_X16Y90:DSP48E2_X16Y113 DSP48E2_X0Y90:DSP48E2_X15Y185 CMACE4_X0Y2:CMACE4_X0Y4 BUFG_GT_SYNC_X0Y60:BUFG_GT_SYNC_X0Y119 BUFG_GT_X0Y96:BUFG_GT_X0Y191 BUFGCTRL_X0Y48:BUFGCTRL_X0Y63 BUFGCTRL_X0Y32:BUFGCTRL_X0Y39 BUFGCE_DIV_X0Y24:BUFGCE_DIV_X0Y31 BUFGCE_DIV_X0Y16:BUFGCE_DIV_X0Y19 BUFGCE_X0Y144:BUFGCE_X0Y191 BUFGCE_X0Y96:BUFGCE_X0Y119 BITSLICE_TX_X0Y48:BITSLICE_TX_X0Y63 BITSLICE_TX_X0Y32:BITSLICE_TX_X0Y39 BITSLICE_RX_TX_X0Y312:BITSLICE_RX_TX_X0Y415 BITSLICE_RX_TX_X0Y208:BITSLICE_RX_TX_X0Y259 BITSLICE_CONTROL_X0Y48:BITSLICE_CONTROL_X0Y63 BITSLICE_CONTROL_X0Y32:BITSLICE_CONTROL_X0Y39 BIAS_X0Y12:BIAS_X0Y15 BIAS_X0Y8:BIAS_X0Y9 SLICE_X166Y240:SLICE_X175Y479 SLICE_X164Y300:SLICE_X165Y419 SLICE_X154Y240:SLICE_X163Y479 SLICE_X152Y300:SLICE_X153Y419 SLICE_X141Y240:SLICE_X151Y479 SLICE_X139Y300:SLICE_X140Y419 SLICE_X126Y240:SLICE_X138Y479 SLICE_X124Y300:SLICE_X125Y419 SLICE_X122Y240:SLICE_X123Y479 SLICE_X117Y360:SLICE_X121Y479 SLICE_X117Y240:SLICE_X121Y299 SLICE_X113Y240:SLICE_X116Y479 SLICE_X111Y300:SLICE_X112Y419 SLICE_X99Y240:SLICE_X110Y479 SLICE_X97Y300:SLICE_X98Y419 SLICE_X87Y240:SLICE_X96Y479 SLICE_X85Y300:SLICE_X86Y419 SLICE_X65Y240:SLICE_X84Y479 SLICE_X63Y300:SLICE_X64Y419 SLICE_X52Y240:SLICE_X62Y479 SLICE_X50Y300:SLICE_X51Y419 SLICE_X39Y240:SLICE_X49Y479 SLICE_X37Y300:SLICE_X38Y419 SLICE_X21Y240:SLICE_X36Y479 SLICE_X19Y300:SLICE_X20Y419 SLICE_X10Y240:SLICE_X18Y479 SLICE_X8Y300:SLICE_X9Y419 SLICE_X0Y240:SLICE_X7Y479 '.
Resolution: add the missing ranges to the parent or child Pblock.
INFO: [Project 1-461] DRC finished with 1 Errors
INFO: [Project 1-462] Please refer to the DRC report (report_drc) for more information.
ERROR: [Vivado_Tcl 4-78] Error(s) found during DRC. Opt_design not run.

Time (s): cpu = 00:00:17 ; elapsed = 00:00:11 . Memory (MB): peak = 6982.004 ; gain = 0.000 ; free physical = 12438 ; free virtual = 29201
INFO: [Common 17-83] Releasing license: Implementation
5 Infos, 0 Warnings, 0 Critical Warnings and 2 Errors encountered.
opt_design failed
opt_design: Time (s): cpu = 00:00:18 ; elapsed = 00:00:12 . Memory (MB): peak = 6982.004 ; gain = 0.000 ; free physical = 12438 ; free virtual = 29201
ERROR: [Common 17-39] 'opt_design' failed due to earlier errors.

    while executing
"opt_design"
    (file "/home/dev/aws-fpga/hdk/cl/examples/cl_dram_hbm_dma/build/scripts/build_level_1_cl.tcl" line 68)

    while executing
"source ${scripts_dir}/build_level_1_cl.tcl"
    ("default" arm line 3)
    invoked from within
"switch $BUILD_FLOW {
  "SynthCL" {
    source ${scripts_dir}/synth_${CL}.tcl
  }

  "ImplCL" {
    source ${scripts_dir}/build_level_1_cl.tcl
  }

  d..."
    (file "build_all.tcl" line 265)
INFO: [Common 17-206] Exiting Vivado at Mon Jun  2 22:00:20 2025...

AWSjoeluc, 2025-06-02:
Hi @dustydude42,

Thank you for the feedback on CL_TEMPLATE. Were you able to follow Vivado's recommendation in the log above? `Resolution: add the missing ranges to the parent or child Pblock.`

It's hard to assist without knowing the Verilog changes made. It's possible that your changes expanded to design into new pblocks that must be added to the constraints.

Please let us know if you have any luck with adding the missing ranges!
- Joe

dustydude42, 2025-06-03:
Hi Joe, thank you for the quick response!

I did try changing the ranges of the parent and child blocks. Here is the original small_shell_cl_pnr_user code block for the  cl_dram_hbm_dma example.

```
###############################################################################
# Child Pblock in SLR1
###############################################################################
########################################
# Pblock
########################################
create_pblock pblock_CL_SLR1

# Complete CRs in SLR1
resize_pblock pblock_CL_SLR1 -add {CLOCKREGION_X0Y4:CLOCKREGION_X3Y7}
resize_pblock pblock_CL_SLR1 -add {CLOCKREGION_X5Y4:CLOCKREGION_X5Y7}
resize_pblock pblock_CL_SLR1 -add {CLOCKREGION_X4Y6:CLOCKREGION_X4Y7}
resize_pblock pblock_CL_SLR1 -add {CLOCKREGION_X4Y4:CLOCKREGION_X4Y4}

# Partial CRs
resize_pblock pblock_CL_SLR1 -add {SLICE_X122Y300:SLICE_X145Y359 \
                                   DSP48E2_X16Y114:DSP48E2_X19Y137 \
                                   RAMB18_X9Y120:RAMB18_X9Y143 \
                                   RAMB36_X9Y60:RAMB36_X9Y71 \
                                   URAM288_X2Y80:URAM288_X2Y95}
resize_pblock pblock_CL_SLR1 -add {RAMB18_X8Y120:RAMB18_X8Y143 \
                                   RAMB36_X8Y60:RAMB36_X8Y71}

set_property SNAPPING_MODE ON [get_pblocks pblock_CL_SLR1]

set_property parent pblock_CL [get_pblocks pblock_CL_SLR1]

########################################
# Module Mapping
########################################
add_cells_to_pblock pblock_CL_SLR1 [get_cells [list WRAPPER/CL/CL_OCL_SLV \
                                                    WRAPPER/CL/CL_SDA_SLV \
                                                    WRAPPER/CL/CL_PCIM_MSTR \
                                                    WRAPPER/CL/CL_DRAM_DMA_AXI_MSTR \
                                                    WRAPPER/CL/CL_DMA_PCIS_SLV/CDC_ASYNC_RST_N_SLR1 \
                                                    WRAPPER/CL/CL_DMA_PCIS_SLV/SLR1_PIPE_RST_N \
                                                    WRAPPER/CL/CL_DMA_PCIS_SLV/AXI4_REG_SLC_PCIS_SLR1 \
                                                    WRAPPER/CL/CL_DMA_PCIS_SLV/AXI4_REG_SLC_DDRA_SLR1 \
                                                    WRAPPER/CL/CL_DMA_PCIS_SLV/AXI4_REG_SLC_DDRB_SLR1 \
                                                    WRAPPER/CL/CL_DMA_PCIS_SLV/AXI4_CROSSBAR]]

###############################################################################
# Grandchild Pblock
# NOTE:
#   This pblock is to help move the PCIS interconnect away from the Shell to
#   avoid routing congestion
###############################################################################
########################################
# Pblock
########################################
create_pblock pblock_CL_SLR1_XBAR

# Complete CRs in SLR1
resize_pblock pblock_CL_SLR1_XBAR -add {CLOCKREGION_X0Y4:CLOCKREGION_X4Y7}

# Partial CRs
set_property parent pblock_CL_SLR1 [get_pblocks pblock_CL_SLR1_XBAR]

########################################
# Module Mapping
########################################
add_cells_to_pblock pblock_CL_SLR1_XBAR [get_cells [list WRAPPER/CL/CL_DMA_PCIS_SLV/AXI4_CROSSBAR ]]
```

The error I am running into is that 
[... 27 lines cut ...]
WARNING: [Vivado 12-180] No cells matched 'WRAPPER/CL/CL_DMA_PCIS_SLV/SLR1_PIPE_RST_N'. [/home/dev/aws-fpga/hdk/cl/examples/cl_dram_hbm_dma/build/constraints/small_shell_cl_pnr_user.xdc:75]
WARNING: [Vivado 12-180] No cells matched 'WRAPPER/CL/CL_DMA_PCIS_SLV/AXI4_REG_SLC_PCIS_SLR1'. [/home/dev/aws-fpga/hdk/cl/examples/cl_dram_hbm_dma/build/constraints/small_shell_cl_pnr_user.xdc:75]
WARNING: [Vivado 12-180] No cells matched 'WRAPPER/CL/CL_DMA_PCIS_SLV/AXI4_REG_SLC_DDRA_SLR1'. [/home/dev/aws-fpga/hdk/cl/examples/cl_dram_hbm_dma/build/constraints/small_shell_cl_pnr_user.xdc:75]
WARNING: [Vivado 12-180] No cells matched 'WRAPPER/CL/CL_DMA_PCIS_SLV/AXI4_REG_SLC_DDRB_SLR1'. [/home/dev/aws-fpga/hdk/cl/examples/cl_dram_hbm_dma/build/constraints/small_shell_cl_pnr_user.xdc:75]
WARNING: [Vivado 12-180] No cells matched 'WRAPPER/CL/CL_DMA_PCIS_SLV/AXI4_CROSSBAR'. [/home/dev/aws-fpga/hdk/cl/examples/cl_dram_hbm_dma/build/constraints/small_shell_cl_pnr_user.xdc:75]
CRITICAL WARNING: [Vivado 12-1433] Expecting a non-empty list of cells to be added to the pblock.  Please verify the correctness of the <cells> argument. [/home/dev/aws-fpga/hdk/cl/examples/cl_dram_hbm_dma/build/constraints/small_shell_cl_pnr_user.xdc:75]
```

In other words, in the following lines

```
add_cells_to_pblock pblock_CL_SLR1 [get_cells [list WRAPPER/CL/CL_OCL_SLV \
                                                    WRAPPER/CL/CL_SDA_SLV \
                                                    WRAPPER/CL/CL_PCIM_MSTR \
                                                    WRAPPER/CL/CL_DRAM_DMA_AXI_MSTR \
                                                    WRAPPER/CL/CL_DMA_PCIS_SLV/CDC_ASYNC_RST_N_SLR1 \
                                                    WRAPPER/CL/CL_DMA_PCIS_SLV/SLR1_PIPE_RST_N \
                                                    WRAPPER/CL/CL_DMA_PCIS_SLV/AXI4_REG_SLC_PCIS_SLR1 \
                                                    WRAPPER/CL/CL_DMA_PCIS_SLV/AXI4_REG_SLC_DDRA_SLR1 \
                                                    WRAPPER/CL/CL_DMA_PCIS_SLV/AXI4_REG_SLC_DDRB_SLR1 \
                                                    WRAPPER/CL/CL_DMA_PCIS_SLV/AXI4_CROSSBAR]]
```
none of the cells get added. Do you have any idea why these cells don't get added to the pblock? Also, could you please explain what the 'WRAPPER/CL' node function is?

Thank you!

AWSjoeluc, 2025-06-09:
Hello!

Thanks for sharing the additional logs. Please try using `pblock_CL` as the parent pblock to yours: `pblock_CL_SLR1_XBAR`. If you continue to experience errors please open an AMD service request [here](https://www.amd.com/en/forms/contact-us/support.html).

As for the `WRAPPER/CL`, that is the hierarchy Vivado exposes to represent our [shell](https://github.com/aws/aws-fpga/blob/f2/hdk/docs/AWS_Shell_Interface_Specification.md). CL should represent the highest level of your Customer Logic design.

Please share your findings from the experiment below!
-Joe