aws/aws-fpga #729: Vitis Platform Install Path

FlyGoat, 2025-06-29:
Hi,

I found an inconsistency of path to Vitis platform files.

For xpfm file setup, the path is hardcoded to:
https://github.com/aws/aws-fpga/blob/ca698e26192f709c2a387569096a3ae3c7a1bee4/vitis_setup.sh#L226

However in later environment setup, it points to a path relative to `XILINX_VITIS`.
https://github.com/aws/aws-fpga/blob/ca698e26192f709c2a387569096a3ae3c7a1bee4/vitis_setup.sh#L280

`XILINX_VITIS` is not guaranteed to be at `/opt/Xilinx/Vitis/${VIVADO_TOOL_VER}` as users may install Vitis to any location. A quick fix would be using `XILINX_VITIS` for `vitis_xpfm_dir` as well. 

I would also argue that installing platform files to Vitis location is rather unwise as it breaks xilinx tools integrity.

FYI for alevo platforms and xilinx embedded platforms, platform files are installed to`/opt/xilinx/platforms`. For Azure FPGA they put xfpm inside their hdk and appened the path to `PLATFORM_REPO_PATHS`.

Thanks

mjthimm, 2025-06-30:
Hi Jiaxun,

Without modifying these paths, are you seeing any failures of the Vitis tools included in the FPGA Developer AMI?

Best,
Max

mjthimm, 2025-06-30:
Hi Jiaxun,

Thank you for letting me know about this!

I will update this in the development kit and we will credit you for this fix when the devkit with it goes live!

Best,
Max

closed 2025-06-30 by mjthimm