aws/aws-fpga #732: How to access (Read/Write)  DDR  from AWS Shell

harishkumar2108, 2025-07-15:
I have a use case of writing and reading data to and from DDR.
Also to access the same data mentioned above and edit it from AWS shell.
How can I do this on F2 instance?
Is there already an example available for that.
If there are no example , Please guide me how to do that.

Currently I am using:
Instance:"i-08f882bdbad8abfb1".
AMI: "AMD_Xilinx_Vivado_2024.1_Developer_AMI_2024_0522_2343_06072024-prod-apl2fcg2okh2s"
 Device for the RTL project: "xcvu47p-fsvh2892-3-e".

I hope this is the right device to use for F2 instance.
I also wanted to know that my RTL part should be same like any other RTL design we do for Xilinx FPGAs, or there should be some changes in it to use it here in AWS platform?

mjthimm, 2025-07-15:
Hi Harish,

Are you trying to write data from the Host to DDR, or from part of your RTL design to DDR?

If you're trying to write to DDR from the host, you'll want to use BAR4 of PF0.

See here for more details on that use case: https://github.com/aws/aws-fpga/tree/f2/hdk/cl/examples/cl_mem_perf#pcis-memory-address-space

Best,
Max

harishkumar2108, 2025-07-16:
Basically I want to write to DDR from Host.
And also read the DDR from RTL.
How can I access from RTL?
Is there any example design?

mjthimm, 2025-07-16:
Hi Harish,

With that link I sent to you, you can see the addresses from the Host side you'll need to write to in order to transfer data to the FPGA's DDR. [Here's a code example of how that happens](https://github.com/aws/aws-fpga/blob/f2/hdk/cl/examples/cl_dram_hbm_dma/software/runtime/test_dram_hbm_dma.c).

The CL examples [`cl_dram_hbm_dma`](https://github.com/aws/aws-fpga/tree/f2/hdk/cl/examples/cl_dram_hbm_dma) and [`cl_mem_perf`](https://github.com/aws/aws-fpga/tree/f2/hdk/cl/examples/cl_mem_perf) are examples that show connections between the main AXI bus in the CL on the FPGA with DDR and HBM, allowing you to write to each.

Best,
Max

harishkumar2108, 2025-07-30:
Thank you Max.
Earlier I had not  sourced sdk_setup.sh.
After Sourcing sdk_setup.sh, When I tried running the "test_dram_hbm_dma" , 
It gave below error:
Am I still missing out something?

[image]

mjthimm, 2025-07-31:
Hi Harish,

We have step-by-step instructions on how to build your DCP here: https://awsdocs-fpga-f2.readthedocs-hosted.com/latest/hdk/README.html

I encourage you to give this guide a read, as answers most common questions about getting your DCP build up and going.

Best,
Max

mjthimm, 2025-08-04:
Hi Harish,

I see the problem here: `Error ... You are not licensed to encrypt IEEE1735 ...`

Are you using the FPGA Developer AMI when you perform these builds? If so, could you provide me the AMI-ID for the version you're using?

With this information, I can dig into this more deeply and get you some answers.

Best,
Max

harishkumar2108, 2025-09-01:
Hello Max.
Do we have to create the AFI in Some other AMI (AMD_Xilinx_Vivado_2024.1_Developer_AMI_2024_0522_2343_06072024-prod-apl2fcg2okh2s) and then use that AFI on another AMI (FPGA Developer AMI) to build and run the example program.

Why am I getting below error for sourcing the "hdk_setup.sh" on your recommended AMI "http://aws.amazon.com/marketplace/pp/prodview-tcl7sjgreh6bq"
?

[image]

can we have a Call to resolve the issue together?

mjthimm, 2025-09-02:
Hi Harish,

One other item I thought of:

When you open the terminal in your DCV environment, did you remember to [run this command at the end of our DCV setup guide](https://github.com/aws/aws-fpga/blob/f2/developer_resources/Amazon_DCV_Setup_Guide.md)?

```
ubuntu@ip-1-2-3-4:~$ source /etc/profile.d/default_module.sh
```

harishkumar2108, 2025-09-16:
Thank you for the Hint @czfpga 

@mjthimm I followed  "https://awsdocs-fpga-f2.readthedocs-hosted.com/latest/hdk/README.html"

The AFI creation is successful.
Loaded the AFI on F2 Instance.
During Validation of AFI using Example Runtime Software, when I executed make command, It's showing below error:

[image]

mjthimm, 2025-09-16:
Hi Harish,

This isn't an error, but a printout of what the `make all` command will be executing when invoked.

Based on what I can see in your screenshot, you ran `make all`, or you built the test binaries individually. You should now be able to run those binaries with `sudo ./test_dram_hbm_dma`, etc.

For the example `cl_dram_hbm_dma`, a DMA engine is required for the examples to work. Have you implemented your own DMA engine in the design before building the AFI? 

Once the XDMA shell is released, this requirement will be lifted and you will be able to use it to automatically insert a DMA engine into your designs.

Best,
Max

harishkumar2108, 2025-09-25:
Hello Max,
How exactly I should use " [the peek/poke strategy to read/write DDR and HBM](https://github.com/aws/aws-fpga/blob/f2/hdk/cl/examples/cl_dram_hbm_dma/software/runtime/test_dram_hbm_dma.c#L237-L242)"
It is not mentioned in the page how exactly to run or use it.
Can you please guide me how to use it, or any document that describes this?
I was referring this page"https://awsdocs-fpga-f2.readthedocs-hosted.com/latest/hdk/README.html"
In this page after running the "make " or "make all" command was throwing error for DMA engine.
But to use the the peak and poke strategy, how shall I proceed?
To start with I have already copied the "test_dram_hbm_dma.c" file to the location "~/hk2025/20250901_aws_fpga_kit/hdk/cl/examples/cl_dram_hbm_dma/software/runtime"
and after that when i run make all, It gives me below error:

ubuntu@ip-172-31-93-194:~/hk2025/20250901_aws_fpga_kit/hdk/cl/examples/cl_dram_hbm_dma/software/runtime$ make all
gcc -DCONFIG_LOGLEVEL=4 -std=gnu11 -g -Wall -Werror -I/home/ubuntu/hk2025/20250901_aws_fpga_kit/sdk/userspace/include -o test_dram_dma_hwsw_cosim test_dram_dma_hwsw_cosim.o test_dram_dma_common.o  -lfpga_mgmt

[image]