aws/aws-fpga #734: Read Write From DDR in F1 instance

samtsaihello, 2025-08-03:
I followed the setting of cl_dram_dma to create my own CL. The log after creating AFI also shows "SUPPORT DDR0", "SUPPORT DDR1", and "SUPPORT DDR2". The parameters DDR_A_PRESENT, DDR_B_PRESENT, DDR_D_PRESENT are all set to 1 in sh_ddr. But when I look at ILA in my own CL, it shows that awready, wready of sh_ddr are both low. sh_cl_ddr_is_ready is also low. I would like to know what else could go wrong so that DRAM can not be used. Thank you!

czfpga, 2025-08-04:
Hi @samtsaihello,

Please check and confirm [`DDR_PRESENT`](https://github.com/aws/aws-fpga/blob/f2/hdk/cl/examples/cl_dram_hbm_dma/design/cl_dram_hbm_dma.sv#L414) is set to True. Also, Shell's `sh_cl_ddr_stat_*` ports are connected to the `sh_ddr` as specified in the document: https://github.com/aws/aws-fpga/tree/f2/hdk/cl/examples/cl_dram_hbm_dma#memory-interface

samtsaihello, 2025-08-04:
Hi @czfpga,

I followed the example cl_dram_dma on the parameter definition and state ports connection. Lib_pipes are also implemented as the example project does. But still the sh_cl_ddr_is_ready signal is not pulled up. I also created an AFI for cl_dram_dma and the DRAM works normally. I wonder what else do I have to pay attention to so that my DDR can be activated. Thank you!!

czfpga, 2025-08-04:
Hi @samtsaihello,

Let's run a sanity check first. Can you run your design through simulation and follow this document (https://github.com/aws/aws-fpga/tree/f1_xdma_shell/hdk/cl/examples/cl_dram_dma#test-changes) to see if the DDRs can be initialized in simulation?