aws/aws-fpga #735: How to add new ip to my project?

pyz-creeper, 2025-08-04:
I've read the link https://github.com/aws/aws-fpga/tree/f2/hdk#cl-example-hierarchy and I'ver read the docs on how to create CL_TEMPLATE https://awsdocs-fpga-f2.readthedocs-hosted.com/latest/hdk/cl/examples/CL-TEMPLATE/README.html.
But I still don't understand how to add a known IP to my project. I know it has to be under hdk/common/ip, but I don't know what file should I add.
Can you guys provide a step-by-step example on how to add IP to a project?

czfpga, 2025-08-05:
Hi @pyz-creeper,

> I assumed this meant I needed to manually copy the mult_gen_0.xci file into that directory, so I did.

Please see my previous response below and you can follow the example code pointed out there to specify the .XCI file location. 

> after that, all you need is to add IP's XCI file to your design, you can find to the example in CL_TEMPLATE: https://github.com/aws/aws-fpga/blob/f2/hdk/cl/examples/CL_TEMPLATE/build/scripts/synth_CL_TEMPLATE.tcl#L41-L81.

Also, to limit variable, consider turning off the IP's core container option, so IP will generate a XCI not XCIX. After that, make sure you generate the output files for the new IP. 

Hope that helps.

czfpga, 2025-08-06:
Thank you for sharing this information and identifying the root cause. We appreciate your feedback and will use it to improve the document.