aws/aws-fpga #740: should we ignore this message?: "WARNING: [Route ...] Router estimated timing not met."

dsw, 2025-08-28:
During the build of your example cl_dram_hbm_dma, I saw the following output.

```
WARNING: [Route 35-328] Router estimated timing not met.
Resolution: For a complete and accurate timing signoff, report_timing_summary must be run after route_design. Alternatively, route_design can be run with the -timing_summary option to enable a complete timing signoff at the end of route_design.
Phase 25 Post Router Timing | Checksum: 208b13fd5
```

And yet I do not find a .VIOLATED file.

```
$ pwd                                                                             
/home/ubuntu/aws-fpga
$ find . -iname "*.VIOLATED"

```

So does that mean that the timing was met despite the warning in the output?  I would call this confusing at best.

dsw, 2025-09-03:
> Hi @dsw <https://github.com/dsw>,
>
> Please share the build log for us to investigate. Meanwhile, check
> ${CL_DIR}/build/checkpoints and see what DCP files are saved in it.

What is the build log exactly?  Is it saved in a particular file?  I wish I
had run the build under nohup and would therefore have a log of the output
for you, but I did not.  I just saw that message go by on the screen and I
copied and pasted it into the bug report.

czfpga, 2025-09-03:
Please see https://github.com/aws/aws-fpga/tree/f2/hdk#step-5-explore-build-artifacts for what is a build log and where it can be found in the CL example directory