aws/aws-fpga #741: build checkpoint *.Developer_CL.tar does not seem to actually be a tar file

dsw, 2025-08-28:
The build process seems to produce a .tar file, and your documentation calls it a "tarball", but the gnu utility "tar" cannot read it and neither can gnu emacs.

```
$ pwd; tar --list 2025_08_27-010013.Developer_CL.tar                            
/home/ubuntu/aws-fpga/hdk/cl/examples/cl_dram_hbm_dma/build/checkpoints         
tar: This does not look like a tar archive                                      
tar: 2025_08_27-010013.Developer_CL.tar: Not found in archive                   
tar: Exiting with failure status due to previous errors                         

```

Is this a tar file?  If it is not, why call it a tar file?  Perhaps just call it something else?  I do not remember who said it, but "If you change the semantics, change the name".

czfpga, 2025-09-03:
Hi @dsw,

Thank you for the feedback. The team will be working on a fix for this.

AWSdirajesw, 2025-09-05:
Hi @dsw,

We are working on renaming "tarball" to "tar file" in the documentation. 

We tested the tar command with a `-f` option in the FPGA developer AMI, and it successfully lists all the contents of the tar file. We recommend trying: `tar --list -f 2025_08_27-010013.Developer_CL.tar` 

Please let us know if you run into any issues. 

Thanks!

DavidLiuGit, 2025-09-05:
@dsw https://man7.org/linux/man-pages/man1/tar.1.html

dsw, 2025-09-05:
I think "tarball" is fine, as that is a term of art.  I just found that tar
--list and emacs both reacted as if it was not a .tar file.

dsw, 2025-09-05:
Ok, I take it you are pointing out that I omitted the '-f' option to tar.

tar -t -f 2025_08_27-010013.Developer_CL.tar

./to_aws/

./to_aws/2025_08_27-010013.manifest.txt

./to_aws/2025_08_27-010013.debug_probes.ltx

./to_aws/2025_08_27-010013.SH_CL_routed.dcp

Tar now does something sensible.  I must call this a bug in tar as it could
say "there is no -f parameter".  However, it is also odd that the installed
emacs cannot parse it as a tar file, as emacs used to be able to do that.
Both tools behaving oddly.

closed 2025-09-05 by dsw