aws/aws-fpga #745: vivado gives opaque error messages: "ERROR: [Synth 8-5809] Error generated from encrypted envelope."

dsw, 2025-09-04:
I have spent hours on this with ChatGPT, so I am not posting this error because I am lazy.  There has got to be a way to get better error messages out of the build process than this followed only by a file and a line number.  How do I get Vivado to tell me the actual error message so that I can debug it?

`ERROR: [Synth 8-5809] Error generated from encrypted envelope.`

dsw, 2025-09-04:
> I've modified my synth scripts to build using the unencrypted version of
> my source files. Just switch the read_verilog line in the synth script to
> read from your unencrypted source directory and not from src_post_enc_dir
> and it'll give you sensible error messages.

Can you please give exact instructions on exactly what to change?  Can you
just send a .diff file or just make this the default and push it to the
repo?  Your scripts and makefiles are so convoluted.

ChrisKjellqvist, 2025-09-04:
I don't work for AWS - I just use it, but this is my .tcl. I would guess that they use an encrypted envelope because when you send a .dcp up to the AWS systems to put it on the F2 instance, you're leaking information of your designs (e.g., module names, wire names) which could be proprietary. This is just my guess.

```
#---- User would replace this section -----

# Reading the .sv and .v files, as proper designs would not require reading
# .vh, nor .inc files
read_verilog -sv [glob ~/<YOUR_CL_DIRECTORY>/design/*v]
```

This code snippet replaces [this part](https://github.com/aws/aws-fpga/blob/9ae615e56c0a7e2b1a3c73c81104da2a1d1d9bdd/hdk/cl/examples/CL_TEMPLATE/build/scripts/synth_CL_TEMPLATE.tcl#L31) of the existing synth script.